Nachtzuster/BirdNET-Pi #289: Feature Request: option to have Daily Charts to have solid background

Chewie9999, 2025-02-19:
Hi everyone,
I have recently updated and the way the Daily Charts are generated have changed.
They now have a transparent background, which could be useful, but it used to be solid green, and was much more readable when using the .PNG with a bot to post to social media.

I also noticed that because of this change, if you choose the "dark" theme, none of the bird names are readable in the web GUI as they are black on black.

Would it be possible to have a setting, either in the GUI or via birdnet.conf, that you can set to chose whether the background is solid or transparent?
Thanks

alexbelgium, 2025-02-20:
I'll convert the css when I have some time. I think I initially did it for the dark mode to decrease the contrast. But it's not that important

Chewie9999, 2025-02-21:
hmm, thanks for the reply.
Your screen looks slight different to mine

[image]

Running version: 33e2cb9e96cd5c52dd6d0337f4814f28fb6b2580

Emmo213, 2025-02-22:
We had other themes planned but they were rejected. Either way there's something off with your build because it's not supposed to be black on black. 

[image]

alexbelgium, 2025-02-22:
Yes indeed my graph layout is from my addon, emmo213's one is the one from this repo ; in any case the display you have show there is something wrong!

Chewie9999, 2025-02-22:
Oh well, it's not a big deal.
I usually run the web gui in "light" (green) theme, and it looks fine.
I just use imagemagick mogrify command to change the alpha to a solid colour before posting to social media.

alexbelgium, 2025-02-22:
Well if this element is not up-to-date it's not sure what else... Have you tried just hitting the upgrade button to see if it solved it?

TF-Ingolstadt, 2025-02-23:
Hello, i had the same issue as Chewie9999 and found a way to fix it:
In BirdNET-Pi/scripts/daily_plot.py section # Set up plot axes and titles (line 87)
Changing facecolour = 'none' to facecolor = '#77C487' does the job for me (at least until the next update).

[image]

[image]

Chewie9999, 2025-02-23:
> Hello, i had the same issue as Chewie9999 and found a way to fix it: In BirdNET-Pi/scripts/daily_plot.py section # Set up plot axes and titles (line 87) Changing facecolour = 'none' to facecolor = '#77C487' does the job for me (at least until the next update).

Thanks!
Yeah, be careful with that though, the next git update may not work!

alexbelgium, 2025-02-23:
The next git update will simply overwrite it so it won't cause issues ; however if you are several to see the same we'll look the code itself to find a more global solution! To be sure, what is you test system to make it easier to troubleshoot?

TF-Ingolstadt, 2025-02-23:
Hi, i am running a Raspberry Pi 4b with the full version of Raspberry Pi OS (64 bit) Bookworm with the Raspberry Pi Desktop. Date 2024-11-19. 
The installation of Birdnet Pi is from 05.01.2025, since then there were some updates.
If you need more or different information i need guidance where to find it and how to read out.

alexbelgium, 2025-02-23:
Thanks! So indeed quite recent and post-implementation of the dark mode plot. I'm not super available in the coming days but I'll keep in mind to check!

alexbelgium, 2025-02-23:
What is the `COLOR_SCHEME` value in your birdnet.conf?

alexbelgium, 2025-02-23:
> Hello, i had the same issue as Chewie9999 and found a way to fix it: In BirdNET-Pi/scripts/daily_plot.py section # Set up plot axes and titles (line 87) Changing facecolour = 'none' to facecolor = '#77C487' does the job for me (at least until the next update).
> 
> [image]
> 
> [image]

BTW regarding your fix, you might as well propose a PR! I can do it for you but it's just so you get proper credit :)

alexbelgium, 2025-03-02:
I've made the PR for you and added you as co-author ;-)

Chewie9999, 2025-03-02:
Thanks everyone!

closed 2025-03-08 by Nachtzuster