Nachtzuster/BirdNET-Pi #295: Species Statistics Circle Chart Not Updating

brownsw77, 2025-02-24:
I have a completely new install of Bookworm and Birdnet-Pi that's been running for three days. 
Most of the functions appear to be working well.
I'm a new Birdnet-Pi user, so I may not be interpreting Species Stat correctly, 
but something appears to be broken in the data feed to Species Statistics on the circle char and the 
bar charts don't make sense either, for example I've only had one Hermit Thrush detection ever and it's got the longest bar length, see below:

[image]

joaldes, 2025-03-04:
I also have this issue.

alexbelgium, 2025-03-04:
Indeed I can replicate too it must be a recent update that led to this bug as it was working fine some months ago

alexbelgium, 2025-03-04:
It's due to a breaking change between plotly 5.24.0 and plotly 6.0.0

alexbelgium, 2025-03-04:
It's due to a breaking change between plotly 5.24.0 and plotly 6.0.0, you can restore function by typing : 
```
$PYTHON_VIRTUAL_ENV /usr/bin/pip3 install plotly==5.24.0
```

alexbelgium, 2025-03-04:
I've pushed a PR that adds compatibility with plotly 6 while also working with previous versions

joaldes, 2025-03-04:
Thanks a ton, @alexbelgium !!!

brownsw77, 2025-03-06:
Thanks, as a new user that knows nothing about the SW architecture, it would have taken me a long time to figure out what update broke this piece.  

 

Steve Brown in Virginia, USA

 

Sent: Tuesday, March 4, 2025 4:36 AM
To: Nachtzuster/BirdNET-Pi ***@***.***>
Cc: Steven Brown ***@***.***>; Author ***@***.***>
Subject: Re: [Nachtzuster/BirdNET-Pi] Species Statistics Circle Chart Not Updating (Issue #295)

 

I've pushed a PR that adds compatibility with plotly 6 while also working with previous versions

—

I've pushed a PR that adds compatibility with plotly 6 while also working with previous versions

closed 2025-03-08 by Nachtzuster