Nachtzuster/BirdNET-Pi #307: rtsps stopped working after updating

ccurtisnz, 2025-03-11:
**Describe the bug**
rtsps stopped working after updating

**To Reproduce**
Steps to reproduce the behavior:
Have RTSPS audio streams

After updating to latest version rtsps feeds stop working.
Logs start recording "Option rw_timeout not found."

**Expected behavior**
rtsps streams to work

**Code or log snippets**

To fix I updated line 36 of birdnet_recording.sh to 

      if [[ "$i" == rtsp://* || "$i" == rtsps://* ]]; then

Then issue was resolved.

Nachtzuster, 2025-03-11:
Thanks for reporting.
Could you check if it is ok now:  Settings->System Controls->Update

ccurtisnz, 2025-03-12:
> Thanks for reporting. Could you check if it is ok now: Settings->System Controls->Update

Updated and still working. Thank you :)

closed 2025-03-12 by ccurtisnz