Nachtzuster/BirdNET-Pi #314: wrong NumPy version?

ClemensGruber, 2025-03-23:
I've installt BirdNet-Pi with the latest bookworm version today and got this error message after startup. Seems something went wrong with the installed NumPy version:

```
17:50:42---[server][INFO] LOADING TF LITE MODEL...
17:50:42---A module that was compiled using NumPy 1.x cannot be run in
17:50:42---NumPy 2.1.3 as it may crash. To support both 1.x and 2.x
17:50:42---versions of NumPy, modules must be compiled with NumPy 2.0.
17:50:42---Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.
17:50:42---If you are a user of the module, the easiest solution will be to
17:50:42---downgrade to 'numpy<2' or try to upgrade the affected module.
17:50:42---We expect that some modules will need time to support NumPy 2.
17:50:42---Traceback (most recent call last):  File "/usr/local/bin/birdnet_analysis.py", line 149, in <module>
17:50:42---    main()
17:50:42---  File "/usr/local/bin/birdnet_analysis.py", line 31, in main
17:50:42---    load_global_model()
17:50:42---  File "BirdNET-Pi/scripts/server.py", line 309, in load_global_model
17:50:42---    INTERPRETER = loadModel()
17:50:42---  File "BirdNET-Pi/scripts/server.py", line 44, in loadModel
17:50:42---    myinterpreter = tflite.Interpreter(model_path=modelpath, num_threads=2)
17:50:42---  File "BirdNET-Pi/birdnet/lib/python3.11/site-packages/tflite_runtime/interpreter.py", line 455, in __init__
17:50:42---    _interpreter_wrapper.CreateWrapperFromFile(
17:50:42---AttributeError: _ARRAY_API not found
17:50:42---ImportError: numpy.core.multiarray failed to import
17:50:42---The above exception was the direct cause of the following exception:
17:50:42---Traceback (most recent call last):
17:50:42---  File "/usr/local/bin/birdnet_analysis.py", line 149, in <module>
17:50:42---    main()
17:50:42---  File "/usr/local/bin/birdnet_analysis.py", line 31, in main
17:50:42---    load_global_model()
17:50:42---  File "BirdNET-Pi/scripts/server.py", line 309, in load_global_model
17:50:42---    INTERPRETER = loadModel()
17:50:42---                  ^^^^^^^^^^^
17:50:42---  File "BirdNET-Pi/scripts/server.py", line 44, in loadModel
17:50:42---    myinterpreter = tflite.Interpreter(model_path=modelpath, num_threads=2)
17:50:42---                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
17:50:42---  File "BirdNET-Pi/birdnet/lib/python3.11/site-packages/tflite_runtime/interpreter.py", line 455, in __init__
17:50:42---    _interpreter_wrapper.CreateWrapperFromFile(
17:50:42---SystemError: <built-in method CreateWrapperFromFile of PyCapsule object at 0x7fac0eb6f0> returned a result with an exception set
```

ClemensGruber, 2025-03-23:
In case you have BirdNET-Pi with the wrong Numpy version allready running use: 

```
source birdnet/bin/activate
pip install numpy==1.26.4  
python -c "import numpy as np; print(np.__version__)"
exit
```

Nachtzuster, 2025-03-24:
Thanks for reporting.

Looks like a bug in the requirements for the tf lite wheel.
I've reverted the streamlit version bump that exposed that latent bug until I do the proper fix.

Nachtzuster, 2025-03-30:
I fixed the wheels to explicitly have numpy<2 as a dependency

closed 2025-03-30 by Nachtzuster

ClemensGruber, 2025-03-30:
Many thanks! Confirm, now it is working!