Nachtzuster/BirdNET-Pi #315: Piwheels plotly-6-.0.1 Hashtags don't match requirements file

doowylloh88, 2025-03-25:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem or your installation here.

**Your build**
The hardware on which BirdNET-Pi is running goes here.

**Code or log snippets**
<details>
<summary>log or code</summary>

```
Paste your 
log output 
or 
code here
```
</details>

Nachtzuster, 2025-03-25:
Looks like a problem with the downloaded wheel maybe.
Could you try again? - might just be a network glitch.

doowylloh88, 2025-03-26:
Fresh re-install worked.  Thank you

evvette, 2025-05-13:
I'm having the same issue with plotly.

Brand new Pi4 8GB
128GB SD disk

Repeated attempts fail at plotly. The calculated hash is different each time. I downloaded plotly directly to my desktop (on same local network) and sha256sum gives the correct hash.

Collecting plotly
  Downloading https://www.piwheels.org/simple/plotly/plotly-6.0.1-py3-none-any.whl (14.8 MB)
     ━━━━━━━━━━━━━━━━━━                       6.7/14.8 MB 82.1 kB/s eta 0:01:39
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    plotly from https://www.piwheels.org/simple/plotly/plotly-6.0.1-py3-none-any.whl#sha256=bb8eec2072350f618c1d73c84747597065a8762af6fefbe1fcf7dcc3b5fa8a1a (from -r ./requirements_custom.txt (line 9)):
        Expected sha256 bb8eec2072350f618c1d73c84747597065a8762af6fefbe1fcf7dcc3b5fa8a1a
             Got        e8d75985e4faddeb437157928978585c6fb2e4bc35b85c40758d9846a298eccc

To gain more control I downloaded newinstaller.sh as a separate step and then removed plotlly from requirements.txt and it finishes without error. But of course it doesn't work.

For last attempt I completely removed BirdNET-Pi, then ran the curl command:
$ curl -s https://raw.githubusercontent.com/Nachtzuster/BirdNET-Pi/main/newinstaller.sh | bash
and it again failed on plotly:
Collecting plotly
  Downloading https://www.piwheels.org/simple/plotly/plotly-6.0.1-py3-none-any.whl (14.8 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━             10.4/14.8 MB 81.4 kB/s eta 0:00:54
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    plotly from https://www.piwheels.org/simple/plotly/plotly-6.0.1-py3-none-any.whl#sha256=bb8eec2072350f618c1d73c84747597065a8762af6fefbe1fcf7dcc3b5fa8a1a (from -r ./requirements_custom.txt (line 9)):
        Expected sha256 bb8eec2072350f618c1d73c84747597065a8762af6fefbe1fcf7dcc3b5fa8a1a
             Got        5d55bf0632dedfcc2c13cac57885408db1df0f45c6fe3ec4d40dc34ba0c95ac8

evvette, 2025-05-13:
Reading my post, I just realized that the file download seems to abort early:

Downloading https://www.piwheels.org/simple/plotly/plotly-6.0.1-py3-none-any.whl (14.8 MB)
━━━━━━━━━━━━━━━━━━ 6.7/14.8 MB 82.1 kB/s eta 0:01:39
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE

Downloading https://www.piwheels.org/simple/plotly/plotly-6.0.1-py3-none-any.whl (14.8 MB)
━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 10.4/14.8 MB 81.4 kB/s eta 0:00:54
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. 

So I used wget to download plotly directly to the pi:

$ wget  https://www.piwheels.org/simple/plotly/plotly-6.0.1-py3-none-any.whl
--2025-05-13 15:18:02--  https://www.piwheels.org/simple/plotly/plotly-6.0.1-py3-none-any.whl
Resolving www.piwheels.org (www.piwheels.org)... 46.235.225.189, 93.93.129.174, 2a00:1098:0:82:1000:3b:1:1, ...
Connecting to www.piwheels.org (www.piwheels.org)|46.235.225.189|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 14805757 (14M)
Saving to: ‘plotly-6.0.1-py3-none-any.whl’

plotly-6.0.1-py3-none-any.whl                   53%[===================================================>                                               ]   7.51M  74.2KB/s    in 98s     

2025-05-13 15:19:41 (78.6 KB/s) - Read error at byte 7871890/14805757 (Error decoding the received TLS packet.). Retrying.

--2025-05-13 15:19:42--  (try: 2)  https://www.piwheels.org/simple/plotly/plotly-6.0.1-py3-none-any.whl
Connecting to www.piwheels.org (www.piwheels.org)|46.235.225.189|:443... connected.
HTTP request sent, awaiting response... 206 Partial Content
Length: 14805757 (14M), 6933867 (6.6M) remaining
Saving to: ‘plotly-6.0.1-py3-none-any.whl’

plotly-6.0.1-py3-none-any.whl                  100%[++++++++++++++++++++++++++++++++++++++++++++++++++++==============================================>]  14.12M  77.6KB/s    in 86s     

2025-05-13 15:21:10 (78.5 KB/s) - ‘plotly-6.0.1-py3-none-any.whl’ saved [14805757/14805757]

birdnetpi:~$ sha256sum plotly-6.0.1-py3-none-any.whl 
bb8eec2072350f618c1d73c84747597065a8762af6fefbe1fcf7dcc3b5fa8a1a  plotly-6.0.1-py3-none-any.whl

A second attempt:
$ wget  https://www.piwheels.org/simple/plotly/plotly-6.0.1-py3-none-any.whl
--2025-05-13 15:27:19--  https://www.piwheels.org/simple/plotly/plotly-6.0.1-py3-none-any.whl
Resolving www.piwheels.org (www.piwheels.org)... 93.93.129.174, 46.235.225.189, 2a00:1098:0:82:1000:3b:1:1, ...
Connecting to www.piwheels.org (www.piwheels.org)|93.93.129.174|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 14805757 (14M)
Saving to: ‘plotly-6.0.1-py3-none-any.whl’

plotly-6.0.1-py3-none-any.whl                   56%[======================================================>                                            ]   7.92M  81.2KB/s    in 1m 44s  

2025-05-13 15:29:04 (77.9 KB/s) - Read error at byte 8303890/14805757 (Error decoding the received TLS packet.). Retrying.

--2025-05-13 15:29:05--  (try: 2)  https://www.piwheels.org/simple/plotly/plotly-6.0.1-py3-none-any.whl
Connecting to www.piwheels.org (www.piwheels.org)|93.93.129.174|:443... connected.
HTTP request sent, awaiting response... 206 Partial Content
Length: 14805757 (14M), 6501867 (6.2M) remaining
Saving to: ‘plotly-6.0.1-py3-none-any.whl’

plotly-6.0.1-py3-none-any.whl                  100%[+++++++++++++++++++++++++++++++++++++++++++++++++++++++===========================================>]  14.12M  76.1KB/s    in 85s     

2025-05-13 15:30:31 (74.7 KB/s) - ‘plotly-6.0.1-py3-none-any.whl’ saved [14805757/14805757]

birdnetpi:~$ sha256sum plotly-6.0.1-py3-none-any.whl 
bb8eec2072350f618c1d73c84747597065a8762af6fefbe1fcf7dcc3b5fa8a1a  plotly-6.0.1-py3-none-any.whl

I haven't found the code that does the actual download, I'm python ignorant!
Could a retry be added to the install script?