Nachtzuster/BirdNET-Pi #328: BirdNET-Pi is running unstable

ClemensGruber, 2025-04-13:
I have set up BirdNET-Pi on a Raspberry Pi 4 with the latest software from here. After around three days, BirdNET-Pi was no longer available on the local network. After a power cycle, the Pi was available again, and I could see on the charts that analysis and bird logging seemed to have continued in the meantime, even without web service or a lost connection.

Now, five days later, the same thing happened again: I could not reach BirdNET-Pi, neither through birdnet.local nor via IP. Again, a power cycle restored access to the web interface. But today I noticed that logging for the past two days is missing! :-/

Is there any log file with information on why this happened or what kind of error may have caused the malfunction?

Any hints to avoid this? A watchdog or scheduled restarts during the night to ensure a more stable system?

ClemensGruber, 2025-04-14:
Unfortunately no indication of anything that went wrong, last entries before getting stuck

```
Apr 11 09:27:42 birdnet python3[3140]: [server][INFO] READING AUDIO DATA...
Apr 11 09:27:42 birdnet python3[3140]: [server][INFO] READING DONE! READ 5 CHUNKS.
Apr 11 09:27:42 birdnet python3[3140]: [server][INFO] ANALYZING AUDIO...
Apr 11 09:27:43 birdnet python3[3140]: [server][INFO] DONE! Time 0.85 SECONDS
Apr 11 09:27:43 birdnet python3[3140]: [server][INFO] 0.0;3.0-('Columba palumbus_Ringeltaube', 0.82780904)
Apr 11 09:27:43 birdnet python3[3140]: [server][INFO] 3.0;6.0-('Human_Human', 0.0)
Apr 11 09:27:43 birdnet python3[3140]: [server][INFO] 6.0;9.0-('Columba palumbus_Ringeltaube', 0.13317466)
Apr 11 09:27:43 birdnet python3[3140]: [server][INFO] 9.0;12.0-('Columba palumbus_Ringeltaube', 0.53632265)
Apr 11 09:27:43 birdnet python3[3140]: [server][INFO] 12.0;15.0-('Human_Human', 0.0)

```

drunkenungulate, 2025-04-14:
For a hang like that I would be looking at the power, or more likely the storage. But power drops can cause corruption on the storage that leads to future crashes as well, so they go hand-in-hand. 

For power:
- Are you using an official power supply from Raspberry Pi? 
- What USB devices, Hats or other devices are connected to the Pi?

You could look for undervoltage warnings in the logs:
```
journalctl --grep voltage
```
Even if they don't correlate in timing with the hangs, it could indicate that the Pi is not getting the power it needs. 

For storage:
Testing on a new SD might be a good idea - preferably a high-endurance type from a good brand, from a reputable dealer.*

You could look for fsck results on boot after the crash to see if there has been anything found:
```
journalctl --grep systemd-fsck
```
Not finding a smoking gun there doesn't mean the card is good. It could be defective, or it might be counterfeit. You could use F3 (on Linux) or h2testw (on Windows) to check the card. 

* If you don't have one it might be a good time to consider an SSD. A small M.2 SATA SSD in a compatible external USB enclosure (or bare adapter board) isn't too expensive. Go NVMe if you think you might want a Pi 5 in the future (since it has a PCIe connector and works with NVMe cards directly through a Hat/board). I'm using SSDs through crappy bare boards that you snap to length, and in nicer Orico enclosures and they all work great. 
Plus, unlike the old days, a Pi with recent firmware can boot directly from a USB-connected drive. No need for the SD at all.

ClemensGruber, 2025-04-15:
No undervoltage or any problem found with systemd-fsck. I'm using the official power supply, not hat just the USB soundcard. Any BirdNET-Pi specific logging beside journalctl?

drunkenungulate, 2025-04-15:
Don't think so. The main birdnet service scripts send log text to stdout. They're running as systemd services, so that output is handled by systemd-journald.  And the dump_logs.sh utility script pulls the log data via journalctl. 

Still, a boot session that ends like you found above is almost certainly caused by a system problem. And causes for problems like that could also create the other, less severe symptom (one or more services failing while the OS continues running). Likely a problem with the SD card.

ClemensGruber, 2025-04-15:
It happened again, hmm, no access to the device, and it stopped without any hint right before the last bird log. Bird recording stopped at around 12 pm. 

I found in the log with `journalctl -b -1 -p warning` this hours before, don't know if it is related: 

```
Apr 15 02:13:24 birdnet avahi-daemon[468]: Host name conflict, retrying with birdnet-2
Apr 15 02:52:22 birdnet kernel: Internal error: SP/PC alignment exception: 000000008a000000 [#1] PREEMPT SMP
Apr 15 02:52:22 birdnet kernel: Modules linked in: cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep vc4 snd_u>
Apr 15 02:52:22 birdnet kernel: CPU: 2 PID: 712 Comm: pulseaudio Tainted: G         C         6.6.74+rpt-rpi-v8 #1  Debian 1:>
Apr 15 02:52:22 birdnet kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
Apr 15 02:52:22 birdnet kernel: pstate: 00000005 (nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Apr 15 02:52:22 birdnet kernel: pc : 0x1
Apr 15 02:52:22 birdnet kernel: lr : __mark_inode_dirty+0x60/0x420
Apr 15 02:52:22 birdnet kernel: sp : ffffffc0815dbbe0
Apr 15 02:52:22 birdnet kernel: x29: ffffffc0815dbbe0 x28: 0000000000e85336 x27: ffffff8047dd0f00
Apr 15 02:52:22 birdnet kernel: x26: 0000000000000000 x25: fffffffe016dd5c0 x24: ffffff8048fadc40
Apr 15 02:52:22 birdnet kernel: x23: ffffffc0815dbd60 x22: ffffff804098d800 x21: ffffff806655d8b0
Apr 15 02:52:22 birdnet kernel: x20: ffffff806655d8b0 x19: 0000000000000001 x18: 0000000000000000
Apr 15 02:52:22 birdnet kernel: x17: 0000000000000000 x16: 0000000000000000 x15: 0000007fe008f9d7
Apr 15 02:52:22 birdnet kernel: x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
Apr 15 02:52:22 birdnet kernel: x11: 0000000000000000 x10: 0000000000000008 x9 : ffffffe833f7d9a4
Apr 15 02:52:22 birdnet kernel: x8 : 0000000000000010 x7 : ffffff8041b424f0 x6 : 000000000d204cdd
Apr 15 02:52:22 birdnet kernel: x5 : 0000000067fdadc6 x4 : 0000000067fdadc6 x3 : 8000000000000000
Apr 15 02:52:22 birdnet kernel: x2 : 0000000000000001 x1 : 0000000000000001 x0 : ffffff806655d8b0
Apr 15 02:52:22 birdnet kernel: Call trace:
Apr 15 02:52:22 birdnet kernel:  0x1
Apr 15 02:52:22 birdnet kernel:  generic_update_time+0x54/0x70
Apr 15 02:52:22 birdnet kernel:  file_update_time+0xa8/0xb8
Apr 15 02:52:22 birdnet kernel:  pipe_write+0x3e8/0x6d8
Apr 15 02:52:22 birdnet kernel:  vfs_write+0x2a4/0x318
Apr 15 02:52:22 birdnet kernel:  ksys_write+0xf0/0x118
Apr 15 02:52:22 birdnet kernel:  __arm64_sys_write+0x24/0x38
Apr 15 02:52:22 birdnet kernel:  invoke_syscall+0x50/0x128
Apr 15 02:52:22 birdnet kernel:  el0_svc_common.constprop.0+0xc8/0xf0
Apr 15 02:52:22 birdnet kernel:  do_el0_svc+0x24/0x38
Apr 15 02:52:22 birdnet kernel:  el0_svc+0x38/0xd0
Apr 15 02:52:22 birdnet kernel:  el0t_64_sync_handler+0x100/0x130
Apr 15 02:52:22 birdnet kernel:  el0t_64_sync+0x190/0x198
Apr 15 02:52:22 birdnet kernel: Code: ???????? ???????? ???????? ???????? (????????)
Apr 15 02:52:22 birdnet kernel: ---[ end trace 0000000000000000 ]---
Apr 15 02:52:23 birdnet pulseaudio[712]: q overrun, queuing locally
Apr 15 02:52:23 birdnet pulseaudio[712]: q overrun, queuing locally
Apr 15 02:52:23 birdnet pulseaudio[712]: q overrun, queuing locally
Apr 15 02:52:23 birdnet pulseaudio[712]: q overrun, queuing locally
Apr 15 02:52:23 birdnet pulseaudio[712]: q overrun, queuing locally
Apr 15 02:52:23 birdnet pulseaudio[712]: q overrun, queuing locally
Apr 15 02:52:23 birdnet pulseaudio[712]: q overrun, queuing locally
```

I think best would be to try it on a complete other system, other SD, other RasPi. Seems to be a local problem, assuming no one else is reporting such unstable systems.

ClemensGruber, 2025-04-16:
Not changed anything yet, again a fozen system with

```
Apr 16 08:12:38 birdnet systemd[1]: Started birdnet_recording.service - BirdNET Recording.
Apr 16 08:12:39 birdnet birdnet_recording.sh[516]: E: [pulseaudio] core-util.c: Failed to create secure directory (/run/user/1000/pu>
Apr 16 08:12:39 birdnet birdnet_recording.sh[596]: E: [autospawn] core-util.c: Failed to create secure directory (/run/user/1000/pul>
Apr 16 08:12:39 birdnet birdnet_recording.sh[596]: W: [autospawn] lock-autospawn.c: Cannot access autospawn lock.
Apr 16 08:12:39 birdnet birdnet_recording.sh[596]: E: [pulseaudio] main.c: Failed to acquire autospawn lock
Apr 16 08:12:40 birdnet birdnet_recording.sh[607]: Failed to create secure directory (/run/user/1000/pulse): No such file or directo>
Apr 16 08:12:40 birdnet birdnet_recording.sh[607]: arecord: main:831: audio open error: No such file or directory
Apr 16 08:12:40 birdnet systemd[1]: birdnet_recording.service: Main process exited, code=exited, status=1/FAILURE
Apr 16 08:12:40 birdnet systemd[1]: birdnet_recording.service: Failed with result 'exit-code'.
Apr 16 08:12:43 birdnet systemd[1]: birdnet_recording.service: Scheduled restart job, restart counter is at 1.
Apr 16 08:12:43 birdnet systemd[1]: Stopped birdnet_recording.service - BirdNET Recording.
Apr 16 08:12:43 birdnet systemd[1]: Started birdnet_recording.service - BirdNET Recording.
Apr 16 08:12:43 birdnet birdnet_recording.sh[662]: E: [pulseaudio] core-util.c: Failed to create secure directory (/run/user/1000/pu>
Apr 16 08:12:43 birdnet birdnet_recording.sh[663]: E: [autospawn] core-util.c: Failed to create secure directory (/run/user/1000/pul>
Apr 16 08:12:43 birdnet birdnet_recording.sh[663]: W: [autospawn] lock-autospawn.c: Cannot access autospawn lock.
Apr 16 08:12:43 birdnet birdnet_recording.sh[663]: E: [pulseaudio] main.c: Failed to acquire autospawn lock
Apr 16 08:12:43 birdnet birdnet_recording.sh[666]: Failed to create secure directory (/run/user/1000/pulse): No such file or directo>
Apr 16 08:12:43 birdnet birdnet_recording.sh[666]: arecord: main:831: audio open error: No such file or directory
Apr 16 08:12:43 birdnet systemd[1]: birdnet_recording.service: Main process exited, code=exited, status=1/FAILURE
Apr 16 08:12:43 birdnet systemd[1]: birdnet_recording.service: Failed with result 'exit-code'.
Apr 16 08:12:46 birdnet systemd[1]: birdnet_recording.service: Scheduled restart job, restart counter is at 2.
Apr 16 08:12:46 birdnet systemd[1]: Stopped birdnet_recording.service - BirdNET Recording.
Apr 16 08:12:46 birdnet systemd[1]: Started birdnet_recording.service - BirdNET Recording.
Apr 16 08:12:48 birdnet birdnet_recording.sh[715]: Recording WAVE '/home/birdnet/BirdSongs/StreamData/%F-birdnet-%H:%M:%S.wav' : Sig>
```

I have now changed the Raspberry Pi first. Same power devide, same SD, same mic / USB-soundcard. Let's see how long it is running ... :-/

julianze, 2025-04-16:
I have seen similar issues with my first and new setup (1 week old, Raspberry Pi 5 4GB, official Power supply and WiFi connected).

I didn't looked deeper so far, but will check details next time.

OutdoorRob, 2025-04-28:
Hi, same issue here: 
Raspberry Pi 3B+ / 32 GB SD Card SanDisk Extreme (SDSQXAF-032G-GN6MA) /official power supply / connected via WiFi
Bookworm / kernel 6.6.74+rpt-rpi-v8 (SMP) aarch64 / BirdNET-Pi version installed on 2025-03-23 / no updates so far

1st occurrence: no connection on 2025-04-03 -> power cycle -> all fine (HTTP access + no record data missing)
2nd occurrence: no connection on 2025-04-13 -> power cycle -> all fine (HTTP access + no record data missing)
Problem haven't occurred a third time until today (knock on wood).

Right now (2025-04-28 / 10 pm) "Tools / System Info" reports a CPU temperature of 58 °C and "Soft temperature limit has occurred" - don't know if this could be a problem (thermal throttling / over heating) !?
I copied the logs via SFTP to my windows machine after the 1st occurrence but haven't had the time to dig through. Is there a tool to read them on windows?

Just trying to get a history for the outside temperature of the location where the Pi is located to see if the days when the issue occurred were really hot ones.

riggo37, 2025-04-30:
Connection problems sometimes caused by Pi's wifi power saving mode. I disabled power saving mode and no more problems.
https://raspberrypi.stackexchange.com/questions/96606/make-iw-wlan0-set-power-save-off-permanent