Nachtzuster/BirdNET-Pi #332: Is there any way can just detect and log results in the database but not keep the sound files.

usersync, 2025-04-21:
Hello,

Thank you for your great work.

I’d like to know if there’s a way to configure BirdNET-Pi to detect only specific species (e.g., Black Redstart and Gray Wagtail) and log the results to the database ​without retaining the audio files. 

Since these are resident species I’m monitoring for daily activity patterns, storing sound files is unnecessary and consumes significant disk space.

While I think I could write a script to periodically purge audio files, but it's not a good solution.

Thank you.

[image]

Nachtzuster, 2025-04-21:
> Hello,
> 
> Thank you for your great work.
> 

You're welcome!

You could try 'Disk Management' to keep the the number of files per species below a set limit - eg. 5000:

[image]

usersync, 2025-04-21:
Great!. thank you~

closed 2025-04-21 by usersync