Nachtzuster/BirdNET-Pi #335: Can't get in to 'Tools'

Jon-Roads, 2025-04-27:
So I've been having a real saga trying to get Birdnet to work. I used the OS 64-bit lite from the imager, and added the Nachtzuster code and it did all the install, and seemed to work fine. I could access the interface from my phone and see the detections no problem, but there were a few misidentifications, and I've been told there is an option to sort that out using the Tools menu. However, when I try and click on Tools, it asks for my username and password, and when I put them in and click submit, it just flashes straight back to the enter username and password screen. No matter how many times I do it it just loops back around.
I've tried lowering shields on my browser, turning security off, everything I can think of, but I still can't get into it. 

So, I've formatted the SD card, and installed the OS 64-bit with the desktop, fired it up, put in the Nachtzuster code again, it did all the install, rebooted, and opened up again on the Pi desktop. I thought doing that I'd be able to get in to the Tools section, but all I've got is the pi desktop, and I can't see the Birdnet interface on there. I can get onto the interface using my phone browser on the local network, but still can't log in to access the Tools menu!

Any suggestions?

glennmckechnie, 2025-04-27:
From the README file...
<pre>
Access

The BirdNET-Pi can be accessed from any web browser on the same network:

    http://birdnetpi.local/ OR your Pi's IP address
    Default Basic Authentication Username: birdnet
    Password is empty by default. Set this in "Tools" > "Settings" > "Advanced Settings" 
</pre>

Unless you specifically changed it , the username / password combo is not your system login one (ssh).

Jon-Roads, 2025-04-27:
Oh thank you thank you!!! You have literally just solved it with that! I'm in! Mate you're a diamond, that's been driving me nuts for days!!!!

closed 2025-05-01 by Nachtzuster