Nachtzuster/BirdNET-Pi #338: Feature Request  - Use wiki commons as photo source (and more prominent bird photos)

Matthias84, 2025-04-28:
Hi, I'm pretty new to the field of birdwatching and have a stronger IT background (but enjoy natuer and gardens as well ;) )

After a few days of playing around with my  Birdnet-PI setup, I noticed that for novices the information and esp. the photos of birds is much less prominent, than I would expect it. Indeed I would love to have a gallery view of species that were detected (recently, per day, ...).

As I expect that flickr might limit the amount of API-calls and the API-key genereation is  a hazzle for most non-tech savy users, I like to also suggest to switch to [Wikipedia commons](https://commons.wikimedia.org/) photo gallery. They can bee lookedup via the wikipedia article references, but also via wikida structured entries.

etmjveg, 2025-06-25:
Adding to Matthias:
I think indeed with Flickr we are on a dead end, seen their banning of free API Keys.

I am  not a coder, just a humble user of all your hard work. Adding pictures/thumbnails on a front view would be a great enhancement, and lower the barrier for more novice users, like my family.

A suggestion of another picture source: I ran into the work done by tommyno, resulted in birdtunes.net (github tommyno / birdtunes ). He sources the pictures directly from birdweather based on the Birdnet-pi id (hey, we all have that done that and are sharing our data there …..). I cannot possibly understand his code, but it seems he found a way to use station id, show the data from birdweather, and very nicely puts the birdweather pictures with it. Would be great tomhave something similar in our birdnet-pi environment.

As said, just ansuggestion, keep up the good work, love this !

Grtz
John

Nachtzuster, 2025-09-20:
Wikipedia can now be selected as an image source.
On new installs, this is enabled by default, on an existing install, you'll have to turn it on in Settings->Bird Photo Source

closed 2025-09-20 by Nachtzuster