Nachtzuster/BirdNET-Pi #341: Error in daily_plot permissions when updating from Mcguirepr89

lasiadora1, 2025-04-29:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
On updating from the latest version of McGuirepr89 BirdNetPi using the instructions in the readme, the permissions when running daily_plot cause the main overview plot to be missing the names of the bird species. The heatmap graph with data shows, but the names are missing from the left.

**To Reproduce**
Steps to reproduce the behavior:
1. Update from latest McGuierpr89 build to Nachtzuster build using the instructions provided
2. Leave running for a period of time
3. Log in to birdnetpi, and the graph has not been created as expected

**Expected behavior**
A clear and concise description of what you expected to happen.
That the graph would have the bird species on a per-row basis

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Additional context**
Add any other context about the problem or your installation here.
The daily_plot script is run as pi and managed via systemctl. If I disable this and run the script locally as pi, the output is correct, but when run via systemctl the output is incorrect as per screenshot.

**Your build**
The hardware on which BirdNET-Pi is running goes here.
RPi-4 running Debian Bullseye

**Code or log snippets**
<details>
The error appears to be in this line:

 f, axs = plt.subplots(1, 2, figsize=(10, height), gridspec_kw=dict(width_ratios=[3, 6]), facecolor=facecolor)

As as by the time you reach the line:

   plot = sns.countplot(y='Com_Name', hue='Com_Name', legend=False, data=df_plt_selection_today,
                         palette=dict(zip(confmax.index, colors)), order=freq_order, ax=axs[0], edgecolor='lightgrey')

axs[0] is not instantiated

(note: I am not a python developer, so apologies for the way I've worded this).
<summary>log or code</summary>

```
Paste your 
log output 
or 
code here
```
</details>

Nachtzuster, 2025-05-01:
Thanks for reporting!

You mention 'permissions', could you expand on that a bit?

I can't  reproduce the issue, unfortunately.
Also could you run some commands figure out what is going on?

`whoami`
`ls -l`
`ls -l homepage/`
`ls -l homepage/static/`
`journalctl -eu chart_viewer.service`

lasiadora1, 2025-05-02:
Hi,

Apologies - I used 'permissions' incorrectly - I was presuming that because daily_plot was run via systemctl it might have differing permissions that me running it at the command line as the same user - I may be wrong in that presumption.

I've run the commands as asked:

**whoami**
pi

**ls -l**
-rw-rw-rw-+ 1 pi pi       73 Apr  8 14:12 analyzing_now.txt
-rw-rw-rw-+ 1 pi pi        0 Apr  9 21:58 apprise.txt
-rw-rw-rw-+ 1 pi pi 22243698 May  2 11:30 BirdDB.txt
drwxrwxrwx+ 8 pi pi     4096 Apr  8 14:36 birdnet
-rw-rw-rw-+ 1 pi pi     7510 Apr 29 09:04 birdnet.conf
drwxrwxrwx+ 2 pi pi     4096 Apr  8 14:12 docs
-rw-rw-rw-+ 1 pi pi        0 Jun 13  2022 exclude_species_list.txt
-rw-rw-rw-+ 1 pi pi      709 Jun 13  2022 firstrun.ini
drwxrwxrwx+ 4 pi pi     4096 Apr 29 09:01 homepage
-rw-rw-rw-+ 1 pi pi 18669725 Apr  9 21:25 HUMAN.txt
-rw-rw-rw-+ 1 pi pi        0 Jun 13  2022 include_species_list.txt
-rw-rw-rw-+ 1 pi pi        0 Jun 27  2022 __init__.py
-rw-rw-rw-+ 1 pi pi    84090 Jun 13  2022 LICENSE
drwxrwxrwx+ 2 pi pi     4096 Apr  9 21:58 model
-rwxrwxrwx+ 1 pi pi     1343 Apr  8 14:12 newinstaller.sh
-rw-rw-rw-+ 1 pi pi    12556 Apr  8 14:12 README.md
-rw-rw-rw-+ 1 pi pi      205 Apr  8 14:33 requirements_custom.txt
-rw-rw-rw-+ 1 pi pi      204 Apr 29 09:01 requirements.txt
drwxrwxrwx+ 5 pi pi     4096 May  2 11:30 scripts
drwxrwxrwx+ 2 pi pi     4096 Apr 29 09:01 templates
drwxrwxrwx+ 2 pi pi     4096 Jul 18  2023 tests
-rw-rw-rw-+ 1 pi pi 72164899 Apr  9 09:10 tflite_runtime-2.11.0-cp39-none-linux_aarch64.whl
-rw-rw-rw-+ 1 pi pi     1957 Apr  8 14:12 version.md

**ls -l homepage/**
drwxrwxrwx+ 2 pi pi  4096 Apr  8 14:12 images
-rw-rw-rw-+ 1 pi pi  1933 Apr  8 14:12 index.php
drwxrwxrwx+ 2 pi pi  4096 Apr 29 09:01 static
-rw-rw-rw-+ 1 pi pi 15334 Apr 29 09:01 style.css
-rw-rw-rw-+ 1 pi pi 19988 Apr 29 09:01 views.php

**ls -l homepage/static/**
-rw-rw-rw-+ 1 pi pi  579759 Mar 16  2023 Chart.bundle.js
-rw-rw-rw-+ 1 pi pi    1929 Mar 16  2023 chartjs-plugin-trendline.min.js
-rw-rw-rw-+ 1 pi pi   24426 Apr 29 09:01 custom-audio-player.js
-rw-rw-rw-+ 1 pi pi   16281 Apr 29 09:01 dark-style.css
-rw-rw-rw-+ 1 pi pi     636 Jul 17  2022 dialog-polyfill.css
-rw-rw-rw-+ 1 pi pi   29167 Jul 17  2022 dialog-polyfill.js
-rw-rw-rw-+ 1 pi pi 5732824 Apr  8 14:12 NotoSansJP-Regular.ttf
-rw-rw-rw-+ 1 pi pi   47484 Apr  8 14:12 NotoSansThai-Regular.ttf
-rw-rw-rw-+ 1 pi pi  125587 Apr  8 14:12 plupload.full.min.js
-rw-rw-rw-+ 1 pi pi  109812 Jun 13  2022 RobotoFlex-Regular.ttf

**journalctl -eu chart_viewer.service**
Apr 27 01:20:00 birdnetpi python3[264377]: empty dataset
Apr 27 01:20:00 birdnetpi python3[264377]: empty dataset
Apr 27 01:20:00 birdnetpi python3[264377]: empty dataset
Apr 27 01:20:00 birdnetpi python3[264377]: empty dataset
Apr 27 01:20:00 birdnetpi python3[264377]: empty dataset
Apr 29 09:05:12 birdnetpi systemd[1]: Stopping BirdNET-Pi Chart Viewer Service...
Apr 29 09:05:12 birdnetpi python3[264377]: empty dataset
Apr 29 09:05:12 birdnetpi systemd[1]: chart_viewer.service: Succeeded.
Apr 29 09:05:12 birdnetpi systemd[1]: Stopped BirdNET-Pi Chart Viewer Service.
Apr 29 09:05:12 birdnetpi systemd[1]: chart_viewer.service: Consumed 15h 11min 4.083s CPU time.
Apr 29 09:05:12 birdnetpi systemd[1]: Started BirdNET-Pi Chart Viewer Service.

Nachtzuster, 2025-05-03:
Thanks!
No no errors in the log, the permissions look fine. hmmm, I wonder what is going on.

Could you run `ps auxwf` when under `systemctl` and report back?

lasiadora1, 2025-05-03:
Hi,

No problems - as below:

USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root           2  0.0  0.0      0     0 ?        S    Apr13   0:07 [kthreadd]
root           3  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [rcu_gp]
root           4  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [rcu_par_gp]
root           8  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [mm_percpu_wq]
root           9  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [rcu_tasks_kthre]
root          10  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [rcu_tasks_rude_]
root          11  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [rcu_tasks_trace]
root          12  0.0  0.0      0     0 ?        S    Apr13  19:49  \_ [ksoftirqd/0]
root          13  0.1  0.0      0     0 ?        I    Apr13  46:31  \_ [rcu_preempt]
root          14  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [migration/0]
root          15  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [cpuhp/0]
root          16  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [cpuhp/1]
root          17  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [migration/1]
root          18  0.0  0.0      0     0 ?        S    Apr13  18:58  \_ [ksoftirqd/1]
root          21  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [cpuhp/2]
root          22  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [migration/2]
root          23  0.0  0.0      0     0 ?        S    Apr13  18:36  \_ [ksoftirqd/2]
root          26  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [cpuhp/3]
root          27  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [migration/3]
root          28  0.0  0.0      0     0 ?        S    Apr13  18:16  \_ [ksoftirqd/3]
root          31  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [kdevtmpfs]
root          32  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [netns]
root          33  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [inet_frag_wq]
root          38  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [kauditd]
root          40  0.0  0.0      0     0 ?        S    Apr13   0:02  \_ [khungtaskd]
root          41  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [oom_reaper]
root          42  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [writeback]
root          43  0.0  0.0      0     0 ?        S    Apr13  14:20  \_ [kcompactd0]
root          63  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [kblockd]
root          64  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [blkcg_punt_bio]
root          65  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [watchdogd]
root          68  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [rpciod]
root          70  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [xprtiod]
root          72  0.0  0.0      0     0 ?        S    Apr13  22:40  \_ [kswapd0]
root          73  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [nfsiod]
root          74  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [kthrotld]
root          75  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [iscsi_eh]
root          76  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [iscsi_conn_clea]
root          77  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [nvme-wq]
root          78  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [nvme-reset-wq]
root          79  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [nvme-delete-wq]
root          80  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [dwc_otg]
root          81  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [DWC Notificatio]
root          82  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [uas]
root          84  0.0  0.0      0     0 ?        S<   Apr13   0:00  \_ [vchiq-slot/0]
root          85  0.0  0.0      0     0 ?        S<   Apr13   0:00  \_ [vchiq-recy/0]
root          86  0.0  0.0      0     0 ?        S<   Apr13   0:00  \_ [vchiq-sync/0]
root          87  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [zswap-shrink]
root          88  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [mmc_complete]
root          92  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [irq/184-usb-001]
root         105  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [mld]
root         106  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [ipv6_addrconf]
root         178  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [vchiq-keep/0]
root         180  0.0  0.0      0     0 ?        S<   Apr13   0:00  \_ [SMIO]
root         194  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [mmal-vchiq]
root         197  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [mmal-vchiq]
root         199  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [mmal-vchiq]
root         202  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [mmal-vchiq]
root         203  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [mmal-vchiq]
root         204  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [mmal-vchiq]
root         205  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [mmal-vchiq]
root         232  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [cfg80211]
root         238  0.0  0.0      0     0 ?        I<   Apr13   0:00  \_ [brcmf_wq/mmc1:0]
root         242  0.0  0.0      0     0 ?        S    Apr13   0:00  \_ [brcmf_wdog/mmc1]
[... cut ...]
root     2147687  0.0  0.0   8756   540 ?        Ss   Apr30   0:43 /usr/sbin/inetd
pi       2147691  0.0  0.0   6560   364 ?        Ss   Apr30   0:00 bash /usr/local/bin/livestream.sh
pi       2147692 30.5  0.5 567316  4784 ?        SLl  Apr30 1369:55  \_ ffmpeg -nostdin -loglevel error -ac 2 -f alsa -i default -acodec libmp3lame -b:a 320k -ac 2 -content_type audio/mpeg -f mp3 icecast://source:birdnetpi@localhost:8000
systemd+ 2519597  0.0  0.6  88104  5904 ?        Ssl  21:00   0:00 /lib/systemd/systemd-timesyncd

Nachtzuster, 2025-08-13:
@nickjquinn Great find! I wonder if it has to do with an older Python module on those installs..
Could you run these commands, and report back?
```
source birdnet/bin/activate
pip list
```

lasiadora1, 2025-08-14:
Many thanks @nickjquinn for the work around - I can confirm it also works for me. The differences between your pip list and mine are:

cachetools	5.2.0
certifi	2022.5.18.1
click	8.1.3
fonttools	4.33.3
importlib-metadata	4.11.4
iniconfig	1.1.1
inotify	0.2.10
ipykernel	6.13.1
ipython	8.4.0
Jinja2	3.1.2
jsonschema	4.6.0
jupyter-client	7.3.4
jupyter-core	4.10.0
kiwisolver	1.4.3
Markdown	3.3.7
matplotlib	3.5.2
nbclient	0.6.4
nbformat	5.4.0
notebook	6.4.12
oauthlib	3.2.0
Pillow	9.1.1
pip	25.0.1
plotly	5.8.2
pluggy	1.0.0
protobuf	3.20.1
psutil	5.9.1
Pygments	2.12.0
pyparsing	3.0.9
PyYAML	6.0
pyzmq	23.1.0
requests	2.28.0
requests-oauthlib	1.3.1
rich	12.4.4
scikit-learn	1.1.1
scipy	1.8.1
setuptools	62.4.0
terminado	0.15.0
traitlets	5.2.2.post1
typing_extensions	4.13.1
validators	0.20.0
watchdog	2.1.9

lasiadora1, 2025-08-14:
Hi, my list was just the ones that were different from @nickjquinn - not a complete list.

Not it matters - I updated matplotlib as you suggested, and that’s fixed it!