Nachtzuster/BirdNET-Pi #344: Feature Request: Support for Korean Bird Names

generalred-kr, 2025-04-30:
Dear BirdNET-Pi Team,

First of all, thank you for this wonderful project. I’m a user from South Korea and have been using BirdNET-Pi to monitor local bird activity with great satisfaction.

However, I noticed that the common names of detected birds are currently displayed in English (or other supported languages), but not in Korean. This limits accessibility for many Korean users, especially those less comfortable with English.

I’d like to kindly request support for Korean bird names in a future update. 

Thank you again for your great work!

Warm regards,  
YounSeong Kim  
Jeonju, South Korea

Nachtzuster, 2025-05-01:
Hi @generalred-kr, I've re-vived an old branch that should make adding languages easier, turns out we have Korean bird names, but just not exposing them on the UI
So Korean support should be coming soon!

Beware though, that due to a design limitation in BirdNET-Pi, changing language should only done at initial configuration (it uses Common name for accounting, unfortunately)

Nachtzuster, 2025-05-10:
@generalred-kr There is Korean support now!

generalred-kr, 2025-05-11:
thanks so much너무 너무 감사드립니다덕분에 행복하게 지내게 되었습니다고맙습니다나의 iPhone에서 보냄

closed 2025-05-11 by Nachtzuster