Nachtzuster/BirdNET-Pi #353: Typo in species name when creating spectrogram

Svardsten53, 2025-05-10:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
In the Nordic countries we use the letters Å, Ä and Ö (å, ä and ö). When the spectrograms are created these are not printed, but instead they are A, A and O (a, a and o). The error was already pointed out in the McGuire version, but no action was taken. Could it be that it is only a simple change that needs to be made to correct the error?

Otherwise, the names are correct.

**To Reproduce**
Steps to reproduce the behavior:
N/A

**Expected behavior**
National letters are printed correctly in species names in spectrograms

**Screenshots**
N/A

**Your build**
Raspberry Pi 4B 4 GB

**Code or log snippets**
N/A
</details>

Nachtzuster, 2025-05-10:
Are you seeing this with a up-to-date version?
There is a change pushed today that might fix this.

Svardsten53, 2025-05-11:
Program updated and problem solved. Thank you!

closed 2025-05-11 by Nachtzuster