Nachtzuster/BirdNET-Pi #354: Lost settings, but still running...

craigusus, 2025-05-10:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
The settings page has no completed fields yet is still running. i.e., info is still getting to my Home Assistant via mqtt etc.

**Expected behavior**
To show correct settings

**Screenshots**
It should be using the dark theme.
5 most recent are missing:

[image]

All this info should be filled in; it took me ages to get correct!

[image]

The running version is missing:

[image]

**Your build**
Raspberry Pi 5

[image]

craigusus, 2025-05-10:
birdnet.conf and apprise.txt all have the correct information in them. But not being shown on the page/site.

alexbelgium, 2025-05-11:
Hi, isn't there something in your startup log ? Are your files permissions fine ?

craigusus, 2025-05-11:
I think it's dying; no detections this morning, no updates in Home Assistant and I know full well there was a Wren in the garden!

Sorry, where is the startup log?

Permissions:
```
birdnetpi:~/BirdNET-Pi$ ls -lah
total 125M
drwxrwxr-x 11 craig craig 4.0K May 11 09:13 .
drwxr-x---  8 craig craig 4.0K Sep 22  2024 ..
-rw-rw-rw-  1 craig craig  128 Mar 11 15:12 apprise.txt
-rw-rw-rw-  1 craig craig 7.8M May 10 21:49 BirdDB.txt
drwxr-xr-x  8 craig craig 4.0K Jun  2  2024 birdnet
-rw-rw-rw-  1 craig craig 9.1K May 10 22:03 birdnet.conf
drwxrwxr-x  2 craig craig 4.0K Jun  2  2024 docs
-rw-rw-rw-  1 craig craig 1.3K Apr 18 22:03 exclude_species_list.txt
-rw-rw-rw-  1 craig craig 1.4K Jun  2  2024 firstrun.ini
-rw-rw-rw-  1 craig craig   31 Jun  2  2024 .flake8
drwxrwxr-x  8 craig craig 4.0K May 10 21:51 .git
drwxrwxr-x  4 craig craig 4.0K Jun  2  2024 .github
-rw-rw-rw-  1 craig craig  990 May 10 21:51 .gitignore
drwxrwxr-x  4 craig craig 4.0K Apr 18 22:08 homepage
-rw-rw-rw-  1 craig craig    0 Jun  2  2024 __init__.py
-rw-rw-rw-  1 craig craig  83K Jun  2  2024 LICENSE
drwxrwxr-x  4 craig craig 4.0K May 10 21:51 model
-rwxrwxrwx  1 craig craig 1.4K Mar 19 13:50 newinstaller.sh
-rw-rw-rw-  1 craig craig  13K Mar 30 20:18 README.md
-rw-rw-rw-  1 craig craig  213 Jun  2  2024 requirements_custom.txt
-rw-rw-rw-  1 craig craig  211 May 10 21:51 requirements.txt
drwxrwxr-x  5 craig craig 4.0K May 10 21:54 scripts
drwxrwxr-x  2 craig craig 4.0K Mar 11 15:13 templates
drwxrwxr-x  2 craig craig 4.0K Jun  2  2024 tests
-rw-rw-rw-  1 craig craig  69M Jun  2  2024 tflite_runtime-2.11.0-cp311-cp311-linux_aarch64.whl
-rw-rw-rw-  1 craig craig  48M Apr 18 22:08 tflite_runtime-2.17.1-cp311-cp311-linux_aarch64.whl
-rw-rw-rw-  1 craig craig 2.0K Jul 15  2024 version.md
```

Getting this when viewing the live log:
```
09:10:03---Traceback (most recent call last):
09:10:03---  File "/usr/local/bin/birdnet_analysis.py", line 149, in <module>
09:10:03---    main()
09:10:03---  File "/usr/local/bin/birdnet_analysis.py", line 31, in main
09:10:03---    load_global_model()
09:10:03---  File "BirdNET-Pi/scripts/server.py", line 308, in load_global_model
09:10:03---    sf_thresh = conf.getfloat('SF_THRESH')
09:10:03---                ^^^^^^^^^^^^^^^^^^^^^^^^^^
09:10:03---  File "/usr/lib/python3.11/configparser.py", line 1322, in get
09:10:03---    return _impl(self._name, option, raw=raw, vars=vars,
09:10:03---           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
09:10:03---  File "/usr/lib/python3.11/configparser.py", line 839, in getfloat
09:10:03---    return self._get_conv(section, option, float, raw=raw, vars=vars,
09:10:03---           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
09:10:03---  File "/usr/lib/python3.11/configparser.py", line 824, in _get_conv
09:10:03---    return self._get(section, conv, option, raw=raw, vars=vars,
09:10:03---           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
09:10:03---  File "/usr/lib/python3.11/configparser.py", line 819, in _get
09:10:03---    return conv(self.get(section, option, **kwargs))
09:10:03---           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
09:10:03---ValueError: could not convert string to float: ''
09:10:06---Traceback (most recent call last):
09:10:06---  File "/usr/local/bin/birdnet_analysis.py", line 149, in <module>
09:10:06---    main()
09:10:06---  File "/usr/local/bin/birdnet_analysis.py", line 31, in main
09:10:06---    load_global_model()
09:10:06---  File "BirdNET-Pi/scripts/server.py", line 308, in load_global_model
09:10:06---    sf_thresh = conf.getfloat('SF_THRESH')
09:10:06---                ^^^^^^^^^^^^^^^^^^^^^^^^^^
09:10:06---  File "/usr/lib/python3.11/configparser.py", line 1322, in get
09:10:06---    return _impl(self._name, option, raw=raw, vars=vars,
09:10:06---           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
09:10:06---  File "/usr/lib/python3.11/configparser.py", line 839, in getfloat
09:10:06---    return self._get_conv(section, option, float, raw=raw, vars=vars,
09:10:06---           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
09:10:06---  File "/usr/lib/python3.11/configparser.py", line 824, in _get_conv
09:10:06---    return self._get(section, conv, option, raw=raw, vars=vars,
09:10:06---           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
09:10:06---  File "/usr/lib/python3.11/configparser.py", line 819, in _get
09:10:06---    return conv(self.get(section, option, **kwargs))
09:10:06---           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
09:10:06---ValueError: could not convert string to float: ''
```

And see this on the Daily Charts:

[image]

alexbelgium, 2025-05-14:
Mmh I wonder if your system is corrupted indeed... Sdcard failing ?

craigusus, 2025-05-14:
Yeah, it finally died. Couldn't even get a backup off of it. SO had to start afresh.
New SanDisk 128GB Extreme Go installed

closed 2025-05-14 by craigusus