Nachtzuster/BirdNET-Pi #355: feature-request：How to use the alternative labels_zh file for displaying Chinese bird names?

CAbing42, 2025-05-11:
I noticed that there are two labels_zh files in the BirdNet-PI project:

One in BirdNet-PI/model/labels_nm/

Another in BirdNet-PI/model/labels_l18n/

The one in labels_l18n contains more complete and commonly used Chinese bird names, while the one in labels_nm has fewer names or less commonly used ones.

Currently, I have set the database language to Chinese, but the bird names are displayed according to the labels_zh in the labels_nm folder.

How can I configure BirdNet-PI to use the labels_zh file from the labels_l18n folder instead?

Thank you!

CAbing42, 2025-05-11:
Yesterday, I initially tried to update the labels_zh file in the _nm folder by myself. After adding more than 1,500 new Chinese bird names, I realized that a more complete version already exists in the l18n folder — it contains over 6,000 Chinese names, far more than what I had added.
I couldn't help but laugh and cry at the same time. Is there any way to directly use that file instead?

Nachtzuster, 2025-05-11:
Directly using the other file won't work, species will be labeled incorrectly if you do.

Using a little script, I've added Chinese labels from the l18n file to the nm file. There are 5033 lines changed, so that should help a lot.

CAbing42, 2025-05-11:
Thank you very much! However, in the new file you provided, many of the bird names are written in Traditional Chinese. I used a script to replace 659 of them with their Simplified Chinese equivalents.
Here is the version after my replacements.
Maybe we could consider having both Traditional and Simplified Chinese as separate language ？

[labels_zh_standardized.txt](https://github.com/user-attachments/files/20149166/labels_zh_standardized.txt)

Nachtzuster, 2025-05-17:
I would not mind having both.

So help me understand,
The current Chinese labels are a mixture of both Traditional and Simplified Chinese?
Your attached [labels_zh_standardized.txt](https://github.com/user-attachments/files/20149166/labels_zh_standardized.txt) is then Simplified only?