Nachtzuster/BirdNET-Pi #362: Species names not displayed in live spectrogram

Svardsten53, 2025-05-26:
In the live spectrogram, species names are not displayed when calls are identified. I seem to recall that this used to work before, and it definitely does in McGuire. Could it have something to do with one of the recent updates involving the species lists?

**To Reproduce**
Steps to reproduce the behavior:
Open "Spectrogram" in the menu.

**Expected behavior**
Species names should be displayed when identified.

**Screenshots**
N/A

**Additional context**
N/A

**Your build**
Raspeberry Pi 4B 4GB, Bookworm

</details>

blogdemoi, 2025-05-26:
Same thing for me, no species names.

Svardsten53, 2025-05-29:
Although I had some real detections, nothing was printed to the console.

[image]
[image]
[image]

Nachtzuster, 2025-06-01:
Hmmm , seems your browser is not requesting the json files for some reason.
Could you switch to this debug branch that adds some Console debugging and report back the console?

```
git remote remove origin
git remote add origin https://github.com/Nachtzuster/BirdNET-Pi.git
git fetch
./scripts/update_birdnet.sh -b debug_spec
```

Svardsten53, 2025-06-02:
Since I have a modified version with a changed menu and Swedish translation, I don't want to mess with it, but I have done a completely new installation from scratch and without any of my own modifications on a test computer. On the regular computer, I could see that species names in the spectrogram are missing in all the browsers (Chrome, Firefox and Edge) I tried in Windows 11, so the error does not seem to be in the browser.

On the new installation on the test computer, printing Swedish species names in the spectrogram worked for 20 minutes, after which nothing was printed. I tried rebooting, updating and uninstalling/installing without any change. When I did "Clear all data", the species names came back for 20 minutes and then disappeared as before. I then tried changing the species names to English in the settings and then it suddenly worked for at least two hours! New test with changing to Finnish species names seemed to work as long as I tested.

My own conclusion is that something stops working when I use Swedish species names and that the error temporarily disappears when clearing all data, which is not an appropriate solution to the problem.

Nachtzuster, 2025-09-20:
There does not seem anyone affected providing any debug info, so closing for now

closed 2025-09-20 by Nachtzuster