Nachtzuster/BirdNET-Pi #370: Installer fails on new Bookworm installation

jddavisuk, 2025-06-13:
On a raspberry pi 4, with a new installation of Bookwork on a blanl SD card the isntaller fails with the following error

Establishing a python virtual environment
+ python3 -m venv birdnet
+ source ./birdnet/bin/activate
++ deactivate nondestructive
++ '[' -n '' ']'
++ '[' -n '' ']'
++ '[' -n /usr/bin/bash -o -n '' ']'
++ hash -r
++ '[' -n '' ']'
++ unset VIRTUAL_ENV
++ unset VIRTUAL_ENV_PROMPT
++ '[' '!' nondestructive = nondestructive ']'
++ VIRTUAL_ENV=/home/pi/BirdNET-Pi/birdnet
++ export VIRTUAL_ENV
++ _OLD_VIRTUAL_PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
++ PATH=/home/pi/BirdNET-Pi/birdnet/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
++ export PATH
++ '[' -n '' ']'
++ '[' -z '' ']'
++ _OLD_VIRTUAL_PS1=
++ PS1='(birdnet) '
++ export PS1
++ VIRTUAL_ENV_PROMPT='(birdnet) '
++ export VIRTUAL_ENV_PROMPT
++ '[' -n /usr/bin/bash -o -n '' ']'
++ hash -r
+ pip3 install -U -r /home/pi/BirdNET-Pi/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.

drunkenungulate, 2025-06-13:
@jddavisuk - What does `uname -a` show on your pi?

Also, are you sure you're installing BirdNET-Pi from this repo? If you used the full install guide,  it links to the older mcguirepr89 version with its outdated script. Check the curl command and make sure it matches this one in the Nachtzuster fork:
```
curl -s https://raw.githubusercontent.com/Nachtzuster/BirdNET-Pi/main/newinstaller.sh | bash
```

After installing Bookworm, did you do a `sudo apt update && sudo apt upgrade`? 
Anything else installed on the system?

On the original mcguirepr89 version there were some people hitting this same error when they somehow ended up with the 32-bit version of Python on a 64-bit system. You could check for this with:
```
apt list | grep installed | grep python3/
```

jddavisuk, 2025-06-14:
I tried again, this time doing the sudo apt update && sudo apt upgrade before installation - it then installed correctly.
Thanks