Nachtzuster/BirdNET-Pi #381: Unable to make changes to settings

phillprice, 2025-07-07:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**

I'm unable to make a change to the settings (specifically in the apprise location but I suspect others the error is

Fatal error: Uncaught TypeError: write: Argument #1 ($stream) must be of type resource, bool given in /home/pi/BirdNET-Pi/scripts/config.php:173 Stack trace:
#0/home/pi/BirdNET-Pi/scripts/config.php(173): write #1 /home/pi/BirdNET-Pi/homepage/views.php(180): include('...) #2 {main} thrown in /home/pi/BirdNET-Pi/scripts/config.php on line 173

**To Reproduce**
Steps to reproduce the behavior:
1. Go to settings
2. Click on apprise area and enter an mqtt url
3. Saw be basic settings 
4. See error

**Expected behavior**
Settings save and file updated. 

**Your build**
i have a new setup where I have a pi 4 that was flashed with rasping and the install script run. 

I then copied over birdnet.conf and the birds.db in the scripts folder from a previous machine.

kpeatt, 2025-07-08:
I had this issue but resolved it by restarting the server after the initial install.

closed 2025-07-09 by Nachtzuster

Nachtzuster, 2025-07-09:
Woops, I should not have closed this issue yet. I read the issue was resolved, but did not look at the poster 😆