Nachtzuster/BirdNET-Pi #399: Species Stats (Streamlit Server) Issue

e-lbarto, 2025-07-28:
[image]

[image]

[image]

[image]

e-lbarto, 2025-07-28:
"Species Stats" not working after setting up a new BirdNET-Pi (Raspi 4, Bookworm 64Lite, latest version). Only Shows "Connecting to Streamlit Server".  Not Showing An Overview of the Daily Detections either. (screenshot 1 and 2) Tried different machines (Raspi4, Raspi 3B+), and even an older version of Birdnet-Pi from the reposiory - same issue. Everything working fine on my other instance from April 2025 (all updates). (screenshot 3 and 4)

e-lbarto, 2025-07-28:
Daily Charts is not available either, although the machine is detecting birds.

[image]

e-lbarto, 2025-07-28:
thanks Sieberkev! tried the command, and now this is what i got back.  chart_viewer.service: Main process exited, code=killed )in the end). no idea what to do.

birdnet:~/BirdNET-Pi$ sudo journalctl -u chart_viewer.service -n 200
Jul 28 20:57:59 birdnet systemd[1]: Started chart_viewer.service - BirdNET-Pi Chart Viewer Service.
Jul 28 20:58:00 birdnet systemd[1]: chart_viewer.service: Main process exited, code=killed, status=4/ILL
Jul 28 20:58:00 birdnet systemd[1]: chart_viewer.service: Failed with result 'signal'.
Jul 28 20:58:00 birdnet systemd[1]: chart_viewer.service: Consumed 1.679s CPU time.
Jul 28 21:00:00 birdnet systemd[1]: chart_viewer.service: Scheduled restart job, restart counter is at 54.
Jul 28 21:00:00 birdnet systemd[1]: Stopped chart_viewer.service - BirdNET-Pi Chart Viewer Service.
Jul 28 21:00:00 birdnet systemd[1]: chart_viewer.service: Consumed 1.679s CPU time.
Jul 28 21:00:00 birdnet systemd[1]: Started chart_viewer.service - BirdNET-Pi Chart Viewer Service.
Jul 28 21:00:02 birdnet systemd[1]: chart_viewer.service: Main process exited, code=killed, status=4/ILL
Jul 28 21:00:02 birdnet systemd[1]: chart_viewer.service: Failed with result 'signal'.
Jul 28 21:00:02 birdnet systemd[1]: chart_viewer.service: Consumed 1.746s CPU time.
Jul 28 21:02:02 birdnet systemd[1]: chart_viewer.service: Scheduled restart job, restart counter is at 55.
Jul 28 21:02:02 birdnet systemd[1]: Stopped chart_viewer.service - BirdNET-Pi Chart Viewer Service.
Jul 28 21:02:02 birdnet systemd[1]: chart_viewer.service: Consumed 1.746s CPU time.
Jul 28 21:02:02 birdnet systemd[1]: Started chart_viewer.service - BirdNET-Pi Chart Viewer Service.
Jul 28 21:02:04 birdnet systemd[1]: chart_viewer.service: Main process exited, code=killed, status=4/ILL
Jul 28 21:02:04 birdnet systemd[1]: chart_viewer.service: Failed with result 'signal'.
Jul 28 21:02:04 birdnet systemd[1]: chart_viewer.service: Consumed 2.051s CPU time.
Jul 28 21:04:04 birdnet systemd[1]: chart_viewer.service: Scheduled restart job, restart counter is at 56.
Jul 28 21:04:04 birdnet systemd[1]: Stopped chart_viewer.service - BirdNET-Pi Chart Viewer Service.
Jul 28 21:04:04 birdnet systemd[1]: chart_viewer.service: Consumed 2.051s CPU time.
Jul 28 21:04:04 birdnet systemd[1]: Started chart_viewer.service - BirdNET-Pi Chart Viewer Service.
Jul 28 21:04:05 birdnet systemd[1]: chart_viewer.service: Main process exited, code=killed, status=4/ILL
Jul 28 21:04:05 birdnet systemd[1]: chart_viewer.service: Failed with result 'signal'.
Jul 28 21:04:05 birdnet systemd[1]: chart_viewer.service: Consumed 1.702s CPU time.
Jul 28 21:06:06 birdnet systemd[1]: chart_viewer.service: Scheduled restart job, restart counter is at 57.
Jul 28 21:06:06 birdnet systemd[1]: Stopped chart_viewer.service - BirdNET-Pi Chart Viewer Service.
Jul 28 21:06:06 birdnet systemd[1]: chart_viewer.service: Consumed 1.702s CPU time.
Jul 28 21:06:06 birdnet systemd[1]: Started chart_viewer.service - BirdNET-Pi Chart Viewer Service.
Jul 28 21:06:07 birdnet systemd[1]: chart_viewer.service: Main process exited, code=killed, status=4/ILL
Jul 28 21:06:07 birdnet systemd[1]: chart_viewer.service: Failed with result 'signal'.
Jul 28 21:06:07 birdnet systemd[1]: chart_viewer.service: Consumed 1.703s CPU time.
Jul 28 21:08:07 birdnet systemd[1]: chart_viewer.service: Scheduled restart job, restart counter is at 58.
Jul 28 21:08:07 birdnet systemd[1]: Stopped chart_viewer.service - BirdNET-Pi Chart Viewer Service.
Jul 28 21:08:07 birdnet systemd[1]: chart_viewer.service: Consumed 1.703s CPU time.
Jul 28 21:08:07 birdnet systemd[1]: Started chart_viewer.service - BirdNET-Pi Chart Viewer Service.
Jul 28 21:08:09 birdnet systemd[1]: chart_viewer.service: Main process exited, code=killed, status=4/ILL

Sieberkev, 2025-07-28:
My bad, it was to (possibly/hopefully) help the maintainer, not us users :D

Sieberkev, 2025-07-28:
On my system it detects python 3.11.2, and the install_helpers.sh downloaded the (correct?) tflite_runtime-2.17.1-cp311-cp311-linux_aarch64.whl + altered the requirements_custom.txt to use that specific one.

ShaAlexander, 2025-07-29:
> cd BirdNET-Pi
> source birdnet/bin/activate
> pip uninstall -y pyarrow
> pip install pyarrow=20.0.0
> deactivate

Little syntax error there, 

`_> pip install pyarrow=20.0.0_
`Must be `> pip install pyarrow=20.0.0
`

jsuelwald, 2025-07-30:
@Sieberkev Thanks, this solved it here as well on my fresh install here.

risliv, 2025-07-30:
@Sieberkev thanks, worked here on fresh install as well!

jimenoma, 2025-08-01:
Thanks very much Sieberkev.
Yesterday i did a fresh instalation after my station was working from 1 year in Spain, Madrid, Navalafuente, BirdNET-Pi Piscis, Station 4960, and with the same problems.
Now works perfect¡¡¡