Nachtzuster/BirdNET-Pi #416: Any interest in a visual redesign of the UI?

jesshannon, 2025-08-05:
I’ve just got my BirdNet-PI up and running. Of course the birds in the garden immediately went silent 😆

One of the first things I noticed is that the UI is quite clunky, especially on mobile. Have there been any efforts to redesign it?

If it was rebuilt built on a templating system then it would be easier to add new themes. It would also be an opportunity to make some things more readily extensible such as alternative image sources.

hacustom, 2025-08-11:
Could you share links to those please?  I would be extremely interested in switching to a more actively maintained fork.

dacracot, 2025-08-14:
> Could you share links to those please? I would be extremely interested in switching to a more actively maintained fork.

https://github.com/dacracot/BirdNET-BarChart

This too, is a complete reimplementation that has no common code nor a migration path without further work.