Nachtzuster/BirdNET-Pi #452: Installation guide needs to be updated.

taucler, 2025-08-30:
Hello all,

I don't know if this should be an issue, feel free to delete. 

I am trying to install BirdNET-PI on a Pi5 4GB. 
I've followed the [Installation Guide](https://github.com/mcguirepr89/BirdNET-Pi/wiki/Installation-Guide) from @mcguirepr89 , making sure I selected a 64-bit Lite version of Bookworm, and using this command to install the system:
`curl -s https://raw.githubusercontent.com/Nachtzuster/BirdNET-Pi/main/newinstaller.sh | bash`

At the end of the installation, the Pi5 does not reboot by itself, and trying to connect to the `http://birdnet.local/` results in a caddy webpage.
[image]

After rebooting the Pi5 (`sudo reboot`), the system seems to work just fine!
So maybe the guide should be updated? I understand that @Nachtzuster cannot do that, but maybe a guide on this repo would be usefull?

Anyway, thanks for the work !

johnmcclumpha, 2025-09-04:
I had the same issue when installing, it seems that after `$HOME/BirdNET-Pi/scripts/install_birdnet.sh` completes, the subsequent reboot isn't occurring

roennieinator, 2025-09-13:
Even after a manual reboot I only see the caddy page. I tried reinstalling and even flashing the SD card again. What else can I do now?

RonnyCHL, 2025-09-14:
Open a terminal, ssh to your BirdNET installation and do the command sudo systemctl restart caddy helps for me.