Nachtzuster/BirdNET-Pi #453: Apprise notifications do not send image.

taucler, 2025-08-31:
**Describe the bug**
I've set up Apprise notifications through a Telegram bot posting on a channel to get notified for each new detections (2h between 2 of the same specie): 

> $imageA $comname ($sciname) was just detected ! %0A🔊 Listen to it: $listenurl %0A[$date - $time - $confidencepct%]

The "Send Test Notification" work just fine, I get the image, the custom notification, and I can even linebreak with %0A (linebreak for Telegram bot).
However, when a real notification is sent (thanks to a detection, not test), no image is sent in the channel, and the linebreak doesn't work anymore.

**Expected behavior**
The image should be sent as it is with the Test Notification.

**Your build**
Pi5 4GB

Emmo213, 2025-08-31:
This is actually two separate things:
1) a bug where special characters get dropped from notifications. This can be easily fixed by base64 encoding the notification string but the repo owner doesn't want to do that so the bug remains. 
2) an enhancement where the test notifications and real notifications use the same logic.

Both are available in my repo if you wanted to pull them into your own code base.

taucler, 2025-08-31:
Ok thanks, I'll try your repo for this. 
What should then be used for linebreak ? %0A or \n ?

louispires, 2025-09-10:
2 days ago, I updated and haven't received any Notifications since then.

I was receiving **Images** for each of my notifications - Telegram

Not even Test notification is working anymore for me...

Emmo213, 2025-09-10:
I'm sorry. My notifications and test notifications are working for me. If it's not working for you then I'd recommend moving back to the main project. 

I'm using <br/> with email for a line break.

Nachtzuster, 2025-09-13:
Update:
Something like this now is possible:
[image]

Test message/real message should work the same now as well

As to why some images cannot be sent, I'm not sure, there seem to be reports over on the Apprise issues page about `Bad Request - Unsupported Parameters`. Can't make much of those though

Nachtzuster, 2025-09-14:
I've pushed some changes that make sending Wikipedia images more reliable.
Make sure to wipe the image cache after updating: `rm scripts/wikipedia.db`

@WesSec please try without adding any extra parameters first and see if that works

WesSec, 2025-09-14:
@Nachtzuster  Thanks a lot, if i put $image on a new row, it seems to work a lot better. I will pull latest and remove the old db and report back :)

closed 2025-09-20 by Nachtzuster