Nachtzuster/BirdNET-Pi #458: Sonograms of Eastern Wood-Pewee not appearing on Species Stats tab

Dot-Dash, 2025-09-06:
On the Species Stats tab, if I select Eastern Wood-Pewee in the dropdown at the top "_Which bird would you like to explore for the dates..."_, on the right-hand side, instead of a sonogram for one of the recordings, I get "**RECORDING NOT AVAILABLE :(**" . 

This consistently happens when selecting any of the available recordings. For example, 
_Eastern_Wood-Pewee-76-2025-08-27-birdnet-18:13:53.mp3_. There are at least 20 recordings in that list. As far as I can tell, all of the display the error message when selected. 

This does not happen when selecting any of the other 40-odd species that have been recorded. For those species, the sonogram(s) are shown as expected. 

I know this seems really random. I tried restarting core services as suggested in the troubleshooting guide. There was no change. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Species Stats
2. From the bird dropdown, select Eastern Wood-Pewee. 
4. See error message **RECORDING NOT AVAILABLE :(**

**Expected behavior**
Sonogram is displayed. 

**Screenshots**

[image]
[image]

**Additional context**
System installed about 10 days ago. I have not Excluded or Whitelisted any Species. Note that the sonogram of the Eastern Wood-Pewee does appear as expected in the _Best Recordings_ tab, and the _Recordings_ tab. 

**Your build**
Raspberry Pi 4 Model B (4GB)
Kernel Version 6.12.34 +rpt-rpi-v8 (SMP) aarch64
Distro Name Debian GNU/Linux 12 (bookworm)

**Code or log snippets**
none

```
Paste your 
log output 
or 
code here
```
</details>

alexbelgium, 2025-09-07:
Then indeed it keeps 10 files per species, starting from the highest confidence, except for the files in the last 7 days which explain the 6 additional files

You should probably set it to something like 200 (which means around 700mo per species if you have an average mp3 of 3.5go)

alexbelgium, 2025-09-08:
Actually, checking if files exists would greatly slow down the code, I guess it was implemented this way on purpose