Nachtzuster/BirdNET-Pi #462: Update removed already configured Notification Body text

craigusus, 2025-09-09:
Last updates have wiped my already configured 'Notification Body' text!

[image]

Also in dark mode you cannot see the text.

[image]

[image]

blogdemoi, 2025-09-09:
The same for me:

[image]

Nachtzuster, 2025-09-09:
@craigusus woops. Thanks for reporting

I push the fixes just now.

You can recover by:

- in birdnet.conf, find `#APPRISE_NOTIFICATION_BODY=` and remove the `#`
- remove the body template file: `rm body.txt`
- update once more

craigusus, 2025-09-09:
@Nachtzuster That seems to be ok now, thank you

closed 2025-09-09 by craigusus