Nachtzuster/BirdNET-Pi #465: Daily Charts are Not Produced

Terry-2112, 2025-09-13:
**To Reproduce**
Start BirdNet-pi as add-on to Home Assistant. 
Nothing unusual.

**Expected behavior**
Provide Daily Charts.

For about the past month my BirdNet-pi has not been producing Daily Charts. This worked perfectly before. I have made the recent updates.  The Daily Charts stopped being produced after one of the updates which I made maybe a month ago.

All the usual services are running and I have restarted Home Assistant several times.

**Screenshots**
See attached screenshot.

Note - "No Charts For 2025-09-13". I have seen similar messages each day for about the last month.

[image]

**Your build**
Home Assistant on Raspberry Pi 5.

Nachtzuster, 2025-09-14:
Could you post the entire page please?

Terry-2112, 2025-09-14:
Yes, of course.

Good news is - I did the most recent update this morning and the daily charts are working perfectly again. See attached.

I don't know why it stopped working and why it started working again. Literally all I did between yesterday and today was the most recent update.

I'm happy it is working again - I love this application. Thanks for all your hard work.

[image]

[image]

alexbelgium, 2025-09-14:
Hi, this is due to the homeassistant addon on which I dynamically merge all open PR from my work in addition to the base code which is vanilla Nachtzuster’s code. There was an issue that is now fixed. However those code are specific to the addon and not included in this repo here.

Terry-2112, 2025-09-14:
Ok, thanks.

I just used the PayPal tab on the Home Assistant Addon Info page to buy a coffee. I'm not sure who that goes to but thanks for the support.

alexbelgium, 2025-09-14:
If it’s in the homeassistant addon it goes to me ! Thanks very much ! The only issue with PayPal donations is that it doesn’t provide me with a way to thank the donator - opportunity that you are providing to me right now. So, thank you again!

Terry-2112, 2025-09-22:
Ok, thanks.